raspberrypi/rpi-imager #1031: [FEATURE]: Warn when selecting large disk

mrled, 2025-04-07:
### Is your feature request related to a problem? Please describe.

When using the Imager on a host with multiple removable disks of both large (like a "hard drive", 500GB or more) and small (like a "thumb drive", 64GB or less) capacities, the user _probably_ wants to write the image to a small drive.

Currently there is an "are you sure" modal that asks the user if they really want to delete all the data on the selected disk, but nothing that indicates the disk size. At least on macOS, this uses some hardware name like "Samsung PSSD T7 Shield" or "Seagate Portable Media", which seems to be set by the manufacturer and is different from the hardware path in /dev and the volume name that is set when the disk is formatted.

With several disks connected and OEM hardware names not always easy to parse, this can lead to formatting the wrong device, and an inconvenient backup restore process.

### Describe the solution you would like to see implemented

Special attention paid to large drives. The specific intervention I'm not as sure of. Some ideas:

- Gray out large disks, and require checking a checkbox to ungray them ("allow selecting large disks")
- A modal upon selecting the disk that calls out the size ("this looks like a large disk, are you sure?")
- Something else?

What counts as a "large disk" would have to be some judgement call, maybe 128 or 256GB. The user should be allowed to select these if they want to, but 

Worth noting that balenaEtcher has this, and it has saved me from making a mistake in the past!

### Describe alternatives you've considered

One alternative is simply paying attention, which apparently I need to practice.

### Additional context

I wiped a large disk this morning, intending to wipe a small one. :(

### Version

1.9.2 (Default)

mrled, 2025-04-19:
Here's what balenaEtcher does, in case it's helpful. It shows all the drives, but puts a yelllow badge next to what it considers "large drives".

It also is a bit easier to parse the storage sizes listed in balenaEtcher. I think this is because they are aligned, so it's almost subconscious to compare them. balenaEtcher also uses GB and TB instead of only GB (2TB for balenaEtcher; 2000GB for Raspberry Pi Imager), but I don't think this would detract from understanding if the numerals were aligned in a column.

[image]

Here's a screenshot from Raspberry Pi Imager for comparison:

[image]

nafanz, 2025-04-19:
I didn't think about it before, but now when I compare it, balenaEtcher really does look more friendly.

tdewey-rpi, 2025-05-28:
I'm not inclined to make this change:

1. Multi-TB sdcards are available in standard retail outlets in the Anglosphere
2. What does 'big' mean? Raspberry Pi 5 with a NVME HAT+ can support large-capacity drives, and those are absolutely expected targets.

Unfortunately, this is a powerful tool, and while I'm inclined to accept suggestion that add guard-rails around non-trivial failure cases (eg, validating SSH keys, etc), I'm not inclined to make changes where the fundamental problem is people not paying attention while performing dangerous operations. After you selected that drive, you would have had _another_ confirmation dialog, explicitly warning you of the destructive nature of the operation you were performing - which you must have accepted to proceed to the write.

My stance, therefore, is similar to that of a certain CHARACTER from Terry Pratchett's Discworld novels:

> “You can't give her that!' she screamed. 'It's not safe!'
IT'S A SWORD, said the Hogfather. THEY'RE NOT MEANT TO BE SAFE.
'She's a child!' shouted Crumley.
IT'S EDUCATIONAL.
'What if she cuts herself?'
THAT WILL BE AN IMPORTANT LESSON.”

I understand this will be frustrating. I struggle to think of anyone who has used a computer in the last 30 years who hasn't been in a similar position (myself included multiple times). I hope your backup strategy has you covered.

closed 2025-05-28 by tdewey-rpi