raspberrypi/rpi-imager #1046: [OS Request]: Add IPFire

mtremer, 2025-05-01:
### Is your feature request related to a problem? Please describe.

On behalf of all of us here at IPFire, I would like to request to add IPFire to the imager.

We have been early adopters of ARM and supported Raspberry Pi from the start. It would be great if people would have a nice and easy way to install IPFire on Raspberry Pi.

The JSON file is hosted here: https://www.ipfire.org/downloads/latest/os-list.json

I will submit the linked form shortly. Please let me know if you require any further information.

Best,
-Michael

P.S. I had to submit this based on the Feature template because there was no option to create a free issue.

### Describe the solution you would like to see implemented

N/A

### Describe alternatives you've considered

N/A

### Additional context

N/A

### Version

1.9.2 (Default)

mtremer, 2025-05-01:
Oops, you are right... I corrected this. I must have gotten too excited after I had the main part done :)

tdewey-rpi, 2026-09-01:
Closing as stale.

closed 2026-09-01 by tdewey-rpi