raspberrypi/rpi-imager #1048: [FEATURE]: Is there a log file written to when using this tool?

judfs, 2025-05-02:
### Is your feature request related to a problem? Please describe.

I recently configured a Pi and deployed it. Now I don't have access to it but I would like to know what image I wrote to it. I still have access to the computer I ran rpi-imager on.

### Describe the solution you would like to see implemented

If there already is a log, document it at https://www.raspberrypi.com/documentation/computers/getting-started.html and/or the readme. 
Otherwise create a local log of actions completed and then document it.

### Describe alternatives you've considered

If there is a goto place to look for gui app logs, I don't know it.

### Additional context

v1.8.3 isn't in the version dropdown...

### Version

1.9.2 (Default)

lurch, 2025-05-10:
I don't know if RPi Imager writes a log file or not, but I **do** know that it caches the last image that you wrote. So if you go through each of the OS menus in RPi Imager, and look for one that says "Cached on your computer", that might be the info you're looking for?

tdewey-rpi, 2025-05-12:
No log file is written by this tool. I'd also be opposed to creating one with anything beyond the barest minimal data - no logging, for example, of customisation parameters. This is particularly relevant in single-PC-multi-tenant environments (eg, a shared flashing computer in education or industrial environments), where logs could be captured by malicious actors.

Thankfully, the mitigation is somewhat straightforward - if your use-case requires this information, a notebook and a pen presents the simplest mitigation. If you're feeling more adventurous, you could look at adding manifest files and mDNS services to your flashed OSes.

Closing as 'Won't Address'.

closed 2025-05-12 by tdewey-rpi

lurch, 2025-05-12:
> if your use-case requires this information, a notebook and a pen presents the simplest mitigation.

https://thepihut.com/products/official-raspberry-pi-notebook
https://thepihut.com/products/official-raspberry-pi-ballpoint-pen

😂

rsweet2, 2025-07-09:
If you launch the imager with the debug option you can get a bunch of additional information while running the application. 

You can do it like this on windows from the command line or etc. 
c:\Program Files\Raspberry Pi Imager>rpi-imager.exe --debug