raspberrypi/rpi-imager #1049: [BUG]: rpi-imager immediatly crashes, failing to create OpenGL context

JulianAssmann, 2025-05-04:
### What happened?

When running rpi-imager on a freshly installed Linux Mint 22.1 Xia, it immediatly crashes because of some OpenGL issue:

```
$ rpi-imager
did not find extension DRI_IMAGE_DRIVER version 1
did not find extension DRI_Mesa version 1
failed to load driver: iris
did not find extension DRI_Mesa version 1
failed to load driver: iris
did not find extension DRI_Mesa version 1
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior QSurfaceFormat::DoubleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
Aborted (core dumped)
```

This is the output regardless of whether I install it using `sudo apt-get rpi-imager` or by downloading and installing the `.deb` file.

### Version

1.9.2 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Linux Mint 22.1 Xia

### Selected OS

None

### Which Raspberry Pi Device are you using?

Raspberry Pi 3B, 3A+, 3B+, and Compute Module 3, 3+

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-05-09:
I wonder if this is a case of missing / incorrect OpenGL drivers for your particular GPU? Raspberry Pi Imager v1.8.5 runs perfectly fine on my Ubuntu 22.04 laptop (which apparently has an Intel Core i7-1165G7 CPU with inbuilt Intel Xe TGL GT2 Graphics).

lurch, 2025-05-10:
See also #515

maxnet, 2025-05-10:
>I wonder if this is a case of missing / incorrect OpenGL drivers for your particular GPU?

The easiest way to diagnose if that is the problem would be to start it with software rendering instead:

```
LIBGL_ALWAYS_SOFTWARE=1 rpi-imager
```

(at least when running Linux, things are more complicated with similar problems on Windows and OS X)

tdewey-rpi, 2025-07-21:
See the workaround in #1111.

Unfortunately, however, I cannot offer support for non-Raspberry Pi OS linux distributions. I'd encourage you to try this against the 1.9.6 release, however.