raspberrypi/rpi-imager #1050: [BUG]:  Error while loading shared libraries: libGLX

i0n-, 2025-05-05:
### What happened?

A bug happened... when run the app inside a Raspberry Lite system!

## Replication process:
- Fresh Raspberry Pi OS Lite instalation.
- $ sudo apt update && sudo apt install rpi-imager
- $ sudo rpi-imager --version

### Version

1.9.2 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Raspberry PI Lite

### Selected OS

any

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
/tmp/.mount_rpi-imdkPUM1/AppRun.wrapped: error while loading shared libraries: libGLX.so.0: cannot open shared object file: No such file or directory
```

i0n-, 2025-05-05:
rpi-imager --cli 

Fail too.

nafanz, 2025-05-06:
> sudo apt update && sudo apt install rpi-imager

I think you missed `sudo apt full-upgrade` or forgot to specify it in the description.
That is, you updated the list of packages, but did not install the available updates themselves (if they are available, of course)
I can't say that this will solve the problem. But I think it's worth a try.