raspberrypi/rpi-imager #1053: [BUG]: Raspberry pi imager return 404 while flashing OS in SD Card #405

jesseab, 2025-05-13:
### What happened?

Impossible to use the imager, keep getting 404 when clicking on next, is there an alternative ?

### Version

1.9.0

### What host operating system were you using?

Windows

### Host OS Version

Windows 11

### Selected OS

Raspberry pi os 64 bits

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-05-13:
If you're getting a 404 error then that (probably) means that Imager isn't able to download the OS image that you selected. Are you _sure_ that you selected "Raspberry pi os 64 bits" as the file appears to be there for me: https://downloads.raspberrypi.com/raspios_arm64/images/raspios_arm64-2025-05-13/

jesseab, 2025-05-13:
Yes its this OS version but I instead downloaded the image directly and then from the imager app i could select and install it no problem. The problem is when trying to download from the app it didnt work.

lurch, 2025-05-13:
Very strange. Do you use some kind of proxy or application-firewall? Doesn't seem to be any obvious reason why your web-browser is able to download the OS image but Imager itself isn't. (especially when Imager presumably **is** able to download https://downloads.raspberrypi.com/os_list_imagingutility_v4.json in order to display the list of available-images.)

closed 2025-05-14 by jesseab