raspberrypi/rpi-imager #1054: [BUG]: Unable to build on ChromeOS Debian 12 Bookworm

birkelbot, 2025-05-25:
### What happened?

I'm trying to build rpi-imager from source so that I can use it on my Chromebook.  I followed the [instructions for Linux](https://github.com/raspberrypi/rpi-imager?tab=readme-ov-file#linux).

I updated appimagecraft.yml as follows:
```
version: 1

project:
  name: org.raspberrypi.rpi-imager
  version: 1.9.2

build:
  cmake:
    source_dir: src/
    # You should also set your Qt6_ROOT here, like so:
    extra_variables:
     - Qt6_ROOT=/home/mbirkel91/Qt/6.7.3/gcc_64
  environment:
    BUILD_TYPE: MinSizeRel

appimage:
  linuxdeploy:
    plugins:
      - qt
    raw_environment:
      QML_SOURCES_PATHS: "\"$PROJECT_ROOT\"/src/qmlcomponents/"
      QMAKE: "/home/mbirkel91/Qt/6.7.3/gcc_64/bin/qmake"
    extra_args: --exclude-library "libwayland-*"
    # environment:
      # Disable Wayland back-end until the dependency problem gets resolved.
      # Essentially, importing libqwayland-generic causes the AppImage dependency walker
      # to find libwayland-client from the build OS, and include that in the AppImage.
      # This is deleterious - because the the runtime OS graphics stack also depends on that
      # library to create native window types. With no strong ABI versioning on that library,
      # you wind up with exotic symbol resolution failures at runtime.
      # See-Also: 
      # EXTRA_PLATFORM_PLUGINS: "libqwayland-egl.so;libqwayland-generic.so"
      # EXTRA_QT_PLUGINS: "waylandcompositor"
```

However, I'm getting the following error when running appimagecraft:

```
...
CMake Error at /usr/share/cmake-3.25/Modules/FindLibLZMA.cmake:103 (set_target_properties):
  set_target_properties called with incorrect number of arguments.
Call Stack (most recent call first):
  /home/mbirkel91/rpi-imager/build/_deps/libarchive-src/CMakeLists.txt:508 (FIND_PACKAGE)
...
-- Configuring incomplete, errors occurred!
See also "/home/mbirkel91/rpi-imager/build/CMakeFiles/CMakeOutput.log".
See also "/home/mbirkel91/rpi-imager/build/CMakeFiles/CMakeError.log".
```

Here are the CMake log files:
[CMakeOutput.log](https://github.com/user-attachments/files/20429448/CMakeOutput.log)
[CMakeError.log](https://github.com/user-attachments/files/20429449/CMakeError.log)

Am I doing something wrong?

### Version

1.9.2 (Default)

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

Debian 12 Bookworm

### Selected OS

N/A

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-05-27:
> `CMake Error at /usr/share/cmake-3.25/Modules/FindLibLZMA.cmake:103 (set_target_properties):
  set_target_properties called with incorrect number of arguments.`

If there's a bug in your system-installed version of a cmake file, then maybe you need to use a newer version of `cmake` ?

lurch, 2025-05-27:
Oh, after doing a quick web-search I see that Debian on ChromeOS is running inside a container. So even if you _do_ manage to build RPi Imager inside a container, I strongly doubt that any software running in the container will be allowed low-level block-write access to any SD cards or USB drives.
See also https://github.com/raspberrypi/rpi-imager/issues?q=is%3Aissue%20state%3Aclosed%20chromeos