raspberrypi/rpi-imager #1055: [BUG]: Settings are not being applied

hansu, 2025-05-25:
### What happened?

I have done some settings like 
- hostname
- enable SSH

The hostname is the default one (raspberrypi) and ssh is not enabled.
Further the chosen password and the default password does not work. 
So a bit useless the result.

Version: 1.9.3
Host OS: Ubuntu 22
Chosen Raspberry Pi OS: Bookworm Lite 64 bit (without desktop environment)

### Version

1.9.2 (Default)

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

Ubuntu 22.04 LTS

### Selected OS

Raspberry Pi OS Lite 64 bit Bookworm without desktop environment

### Which Raspberry Pi Device are you using?

Raspberry Pi 3B, 3A+, 3B+, and Compute Module 3, 3+

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
Not loading module "atk-bridge": The functionality is provided by GTK natively. Please try to not load it.
Qt: Session management error: Could not open network socket
Could not attach Keys property to:  OptionsPopup_QMLTYPE_35(0x560da5d2fd00)  is not an Item

(rpi-imager:168057): GLib-GIO-WARNING **: 08:33:46.612: GDBus.Error:org.freedesktop.portal.Error.NotAllowed: This call is not available inside the sandbox
Custom config.txt entries: ""
Custom cmdline.txt entries: ""
Custom firstuse.sh: "#!/bin/bash\n\nset +e\n\nCURRENT_HOSTNAME=`cat /etc/hostname | tr -d \" \\t\\n\\r\"`\nif [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then\n   /usr/lib/raspberrypi-sys-mods/imager_custom set_hostname raspberrypi\nelse\n   echo raspberrypi >/etc/hostname\n   sed -i \"s/127.0.1.1.*$CURRENT_HOSTNAME/127.0.1.1\\traspberrypi/g\" /etc/hosts\nfi\nFIRSTUSER=`getent passwd 1000 | cut -d: -f1`\nFIRSTUSERHOME=`getent passwd 1000 | cut -d: -f6`\nif [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then\n   /usr/lib/raspberrypi-sys-mods/imager_custom enable_ssh\nelse\n   systemctl enable ssh\nfi\nif [ -f /usr/lib/userconf-pi/userconf ]; then\n   /usr/lib/userconf-pi/userconf 'pi' '$5$oE58qRkanU$N14x0ot045669cF2b4QizvlIpYYWX5MUk/EtkT5MVlD'\nelse\n   echo \"$FIRSTUSER:\"'$5$oE58qRkanU$N14x0ot045669cF2b4QizvlIpYYWX5MUk/EtkT5MVlD' | chpasswd -e\n   if [ \"$FIRSTUSER\" != \"pi\" ]; then\n      usermod -l \"pi\" \"$FIRSTUSER\"\n      usermod -m -d \"/home/pi\" \"pi\"\n      groupmod -n \"pi\" \"$FIRSTUSER\"\n      if grep -q \"^autologin-user=\" /etc/lightdm/lightdm.conf ; then\n         sed /etc/lightdm/lightdm.conf -i -e \"s/^autologin-user=.*/autologin-user=pi/\"\n      fi\n      if [ -f /etc/systemd/system/getty@tty1.service.d/autologin.conf ]; then\n         sed /etc/systemd/system/getty@tty1.service.d/autologin.conf -i -e \"s/$FIRSTUSER/pi/\"\n      fi\n      if [ -f /etc/sudoers.d/010_pi-nopasswd ]; then\n         sed -i \"s/^$FIRSTUSER /pi /\" /etc/sudoers.d/010_pi-nopasswd\n      fi\n   fi\nfi\nrm -f /boot/firstrun.sh\nsed -i 's| systemd.run.*||g' /boot/cmdline.txt\nexit 0\n"
Cloudinit: "hostname: raspberrypi\nmanage_etc_hosts: true\npackages:\n- avahi-daemon\napt:\n  conf: |\n    Acquire {\n      Check-Date \"false\";\n    };\n\nusers:\n- name: pi\n  groups: users,adm,dialout,audio,netdev,video,plugdev,cdrom,games,input,gpio,spi,i2c,render,sudo\n  shell: /bin/bash\n  lock_passwd: false\n  passwd: $5$oE58qRkanU$N14x0ot045669cF2b4QizvlIpYYWX5MUk/EtkT5MVlD\n\nssh_pwauth: true\n\n"
qt.qml.context: qrc:/main.qml:1234:9 Parameter "settings" is not declared. Injection of parameters into signal handlers is deprecated. Use JavaScript functions with formal parameters instead.
Custom config.txt entries: ""
Custom cmdline.txt entries: ""
Custom firstuse.sh: "#!/bin/bash\n\nset +e\n\nCURRENT_HOSTNAME=`cat /etc/hostname | tr -d \" \\t\\n\\r\"`\nif [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then\n   /usr/lib/raspberrypi-sys-mods/imager_custom set_hostname raspberrypi\nelse\n   echo raspberrypi >/etc/hostname\n   sed -i \"s/127.0.1.1.*$CURRENT_HOSTNAME/127.0.1.1\\traspberrypi/g\" /etc/hosts\nfi\nFIRSTUSER=`getent passwd 1000 | cut -d: -f1`\nFIRSTUSERHOME=`getent passwd 1000 | cut -d: -f6`\nif [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then\n   /usr/lib/raspberrypi-sys-mods/imager_custom enable_ssh\nelse\n   systemctl enable ssh\nfi\nif [ -f /usr/lib/userconf-pi/userconf ]; then\n   /usr/lib/userconf-pi/userconf 'pi' '$5$KcGA4X3T2O$02Fn4l6GpSI4ZYOqxROreHm2tBPLt6ipR0CUD3pdaA8'\nelse\n   echo \"$FIRSTUSER:\"'$5$KcGA4X3T2O$02Fn4l6GpSI4ZYOqxROreHm2tBPLt6ipR0CUD3pdaA8' | chpasswd -e\n   if [ \"$FIRSTUSER\" != \"pi\" ]; then\n      usermod -l \"pi\" \"$FIRSTUSER\"\n      usermod -m -d \"/home/pi\" \"pi\"\n      groupmod -n \"pi\" \"$FIRSTUSER\"\n      if grep -q \"^autologin-user=\" /etc/lightdm/lightdm.conf ; then\n         sed /etc/lightdm/lightdm.conf -i -e \"s/^autologin-user=.*/autologin-user=pi/\"\n      fi\n      if [ -f /etc/systemd/system/getty@tty1.service.d/autologin.conf ]; then\n         sed /etc/systemd/system/getty@tty1.service.d/autologin.conf -i -e \"s/$FIRSTUSER/pi/\"\n      fi\n      if [ -f /etc/sudoers.d/010_pi-nopasswd ]; then\n         sed -i \"s/^$FIRSTUSER /pi /\" /etc/sudoers.d/010_pi-nopasswd\n      fi\n   fi\nfi\nrm -f /boot/firstrun.sh\nsed -i 's| systemd.run.*||g' /boot/cmdline.txt\nexit 0\n"
Cloudinit: "hostname: raspberrypi\nmanage_etc_hosts: true\npackages:\n- avahi-daemon\napt:\n  conf: |\n    Acquire {\n      Check-Date \"false\";\n    };\n\nusers:\n- name: pi\n  groups: users,adm,dialout,audio,netdev,video,plugdev,cdrom,games,input,gpio,spi,i2c,render,sudo\n  shell: /bin/bash\n  lock_passwd: false\n  passwd: $5$KcGA4X3T2O$02Fn4l6GpSI4ZYOqxROreHm2tBPLt6ipR0CUD3pdaA8\n\nssh_pwauth: true\n\n"
Unmounting: "/dev/sde"
Unmounted "/org/freedesktop/UDisks2/block_devices/sde2" successfully
Unmounted "/org/freedesktop/UDisks2/block_devices/sde1" successfully
BLKDISCARD not supported
Zeroing out first and last MB of drive
Done zeroing out start and end of drive. Took 0 seconds
```

hansu, 2025-05-25:
Seems to be only a problem with German locales. I tried that twice without success. Now I did that with en_US to have an english log and it works. Weird.

lurch, 2025-05-27:
> I tried it several times with different settings and I got it working once. But after that one it didn't work again...

It **may** be that you're simply not waiting long enough for Raspberry Pi OS to do its "first boot tasks":
* resize the root partition 
* resize the root filesystem (which my take quite a while with a very large or very slow SD card)
* generate SSH host keys
* etc.

and so maybe you're trying to connect over SSH before Raspberry Pi OS has actually started the SSH server?

tdewey-rpi, 2025-06-09:
The logs suggest you didn't save the settings. The line starting 'Custom firstrun.sh:' is issued by the command that sets the customisation you've asked for.

In the logs provided, you've not set a different hostname in either call, nor is there any sign that you've enabled SSH.

Could you provide a click-by-click walkthrough of the process you've followed? Does the customisation show up on the dialog?

MontesDesigns, 2025-06-18:
This may be because of an existing issue https://github.com/raspberrypi/rpi-imager/issues/1067

closed 2025-07-21 by tdewey-rpi