raspberrypi/rpi-imager #1057: [BUG]: FullPageOS for Raspberry Pi 5 is not compatible at boot

amah853, 2025-05-26:
### What happened?

**Issue**
When selecting Raspberry Pi 5 in Raspberry Pi Imager and choosing FullPageOS, the written SD card fails to boot and shows the error (not verbatim, it was something along the lines of...):

“The selected image is not compatible with Raspberry Pi 5”

This happens even though Raspberry Pi 5 was explicitly selected in the device menu.

**To Reproduce**

Open the latest version of Raspberry Pi Imager

Choose Raspberry Pi 5 as the target device

Select FullPageOS under specific purpose -> digital signage

Write the image to the SD card

Insert the SD card into Raspberry Pi 5 and boot

**System**

Latest version of Raspberry Pi Imager

Windows 11

Raspberry Pi model: Raspberry Pi 5 (8GB RAM)

### Version

1.9.2 (Default)

### What host operating system were you using?

Windows

### Host OS Version

the latest version of windows 11

### Selected OS

FullPageOS

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-05-27:
This is something that you'll have to chase up with FullPageOS (i.e. ask them to fix their image), it's not something that Raspberry Pi Imager can fix.

Ahhh, I did a quick search and found https://github.com/guysoft/FullPageOS/issues/591

amah853, 2025-05-28:
Tysm!

tdewey-rpi, 2025-06-02:
@amah853 Thanks for the report - unfortunately this is one for guysoft, as pointed out by @lurch.

I'll follow up on this outside of GH issues.

Closing as this isn't an actual issue with `rpi-imager`, but with a third party OS.

closed 2025-06-02 by tdewey-rpi