raspberrypi/rpi-imager #1058: [BUG]: Backslash in SSID generates wrong NetworkManager connection

A1rPun, 2025-05-27:
### What happened?

My network has a backslash in the SSID like this: `\o_example\`.

Rpi_imager correctly detected the SSID and password (in advanced settings) on Windows 10 but it didn't escape the backslashes automatically.

After manually editing `rootfs/etc/NetworkManager/system-connections/iforgotsorry.nmconnection` from:
```
\o_example\
```
to:
```
\\o_example\\
```
It worked and I could continue to work on my project.

### Version

1.9.2 (Default)

### What host operating system were you using?

Windows

### Host OS Version

10

### Selected OS

Lite

### Which Raspberry Pi Device are you using?

Raspberry Pi Zero, Zero W, Zero WH

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

A1rPun, 2025-05-27:
Maybe a relevant issue --> #486

lurch, 2025-05-27:
> Maybe a relevant issue --> [#486](https://github.com/raspberrypi/rpi-imager/issues/486)

Hmmm, looking at the commits that the other issue links to, perhaps we just need to add additional escaping to https://github.com/raspberrypi/rpi-imager/blob/qml/src/OptionsPopup.qml#L335

closed 2025-09-20 by A1rPun