raspberrypi/rpi-imager #1064: [FEATURE]: Recommend Raspberry Pi OS Lite for Pi Zero 2 W in Imager

rd4com, 2025-06-03:
### Is your feature request related to a problem? Please describe.

    Hello, recently received and setup an Raspberry Pi Zero 2 W using Raspberry Pi Imager.
    I initially selected the default Raspberry Pi OS (64-bit).
    
    All great, ssh into it and very slow (many seconds lag) !
    Did not knew that this default recommended OS came with an full desktop.
    
    So did some search engine research, and it turns out there is the lite version of the OS in the menu.
    
    After flashing the light version, everything is great, and ram usage is low. Completely happy 🥳 .
    
    I think it would be a great idea to recommend lite for low ram pi's such as zero 2W.
    
    This small UI tweak would help users choose the most appropriate OS variant for their hardware and avoid a poor first experience.

    Thanks for the great work on the Imager and Raspberry Pi ecosystem!
    

### Describe the solution you would like to see implemented

 I think it would be a great idea to recommend lite for low ram pi's such as zero 2W.

### Describe alternatives you've considered

Nothing special

### Additional context

Nothing special

### Version

1.8.5

lurch, 2025-06-03:
Unfortunately the `tags` and `devices` fields in https://downloads.raspberrypi.com/os_list_imagingutility_v4.json (which is how Imager decides which OSes are compatible with which devices) only includes information about the [CPU type](https://www.raspberrypi.com/documentation/computers/processors.html), and doesn't contain any information about the RAM amounts on different boards.

tdewey-rpi, 2025-06-04:
Thanks for the suggestion @rd4com 

We're actively improving the desktop experience on Raspberry Pi Zero 2 W-class hardware, but I take your point on recommending the `headless` experience.

I expect this sort of dynamic re-sorting of the OS list will only come with the next major UI revision for rpi-imager