raspberrypi/rpi-imager #1067: [BUG]: Wi-Fi config via Raspberry Pi Imager v1.9.4 fails on Raspberry Pi Zero 2 W with 64/32-bit Lite image

nivaluna-py, 2025-06-05:
### What happened?

When using Raspberry Pi Imager v1.9.4 to flash Raspberry Pi OS Lite (64/32-bit, Bookworm) for a Raspberry Pi Zero 2 W, and setting Wi-Fi credentials via the Imager UI, the device boots with:

* Wi-Fi disabled (state: DOWN)
* RFKill blocking wlan0

Note: Using Imager v1.8.5 works perfect!

### Version

1.9.4 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Windows 10 Pro, Version 2009, OS Build 19045.5854

### Selected OS

Raspberry Pi OS Bookworm 64-bit Lite

### Which Raspberry Pi Device are you using?

Raspberry Pi Zero 2 W

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

tdewey-rpi, 2025-06-05:
Thanks for the report, @nivaluna-py 

Could you provide some more detail?

> the device boots with:
>
> Wi-Fi disabled (state: DOWN)
> RFKill blocking wlan0

This is the default state, prior to the image customisation steps running.

How did you confirm this state?

Did you wait for the online resize to complete?

b-morgan, 2025-06-06:
I repeated the experiment using the Raspberry Pi OS (other), Raspberry Pi OS Lite (64-bit) image with the same results, the image written using 1.9.4 does not connect to the WiFi network. I have a monitor, keyboard (and mouse) attached and can easily attach an ethernet cable.

Delta0001, 2025-06-06:
Ran into this last night. Also happens with the Raspberry Pi Zero W.

Steps to reproduce:

1. Set up an image for the Raspberry Pi Zero W.
2. Configure the image with Wifi (including country code). 
3. I also enabled SSH
4. Boot into PI. The wifi will not be properly configured and you will see this when running `rfkill` list
```
0: hci0: Bluetooth
Soft blocked: no
Hard blocked: no
1: phy0: Wireless LAN
Soft blocked: yes
Hard blocked: no
```

5. Setting the country code in raspi-config will change Soft blocked to ``no`` but the wifi continues to not work.

b-morgan, 2025-06-07:
Workarounds are great but the simplest one is using imager 1.8.5 to write the image instead of 1.9.4 until this bug is fixed.

nivaluna-py, 2025-06-07:
Sorry for incomplete issue @tdewey-rpi, it looks like @Delta0001 has the same issue though. Thanks!

Cesarpi2, 2025-06-08:
Rasppberry Pi Imager 1.9.4 does not work with Windows10 pro 32 bits, the program is installed but does not work, inserting and installing version 1.8.5 - it works perfectly, can any friend help? Thanks.

b-morgan, 2025-06-08:
@Cesarpi2, this topic is about issues with the image produced by the 1.9.4 imager. Your issue is about the imager itself and should be a separate issue. That way you can track its resolution separately.

The workaround for both issues is to use the 1.8.5 image.

denicad1, 2025-06-13:
> I'm running Windows 11 Pro. I've never been able to get Wi-Fi to connect automatically on startup with any of my Raspberry Pis (3B, Zero 2 W, or Pi 1), even when configuring Wi-Fi settings through the Raspberry Pi Imager.
> 
> After creating an image for the Zero 2 W, I noticed something in the firstrun.sh script on line 42
> 
> `if [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan 'somessid' 'randomstring' 'US
> 
> '` There is a newline after country code. Removing this new line allowed my PIs to connect on first startup.

THIS WORKED!!!!!

gsantner, 2025-06-13:
Tried now 3 solutions (/boot/wpa_supplicant.conf, custom.toml and imager->firstrun.sh) and none works.
Can it please be made again back to the point where manually creating `/boot/wpa_supplicant.conf` and `/boot/ssh` was a easy working way without a kind of software lockin (raspberry imager).
It was just enough and working to get a raspberry headless setup ready to connect to.

imager_1.9.4_amd64.deb seems not to include newline bug, but still it not works, raspberry not comes up in wifi.

[image]

Have multiple sd cards, card readers and Pi Zero W (1) here - no success.

gsantner, 2025-06-13:
Tried now 3 solutions (/boot/wpa_supplicant.conf, custom.toml and imager->firstrun.sh) and none works.
Can it please be made again back to the point where manually creating `/boot/wpa_supplicant.conf` and `/boot/ssh` was just working and without a kind of software lockin (raspberry imager)?
Headless setup is a bit difficult without wifi, especially when there is no normal HDMI connector either.

imager_1.9.4_amd64.deb seems not to include newline bug, but still it not works, raspberry not comes up in wifi.

[image]

Have multiple sd cards, card readers and Pi Zero W (1) here - no success.

Jebble, 2025-06-14:
Ended up here after trying for hours to get a new Pi Zero 2W to connect to Wi-Fi in headless mode. Pi OS Lite Bullseye was working perfectly fine with `wap_supplicant.conf` but Bookworm just wouldn't connect to the Wi-Fi.

Started with the advise of @b-morgan to use 1.8.5 which also did not work for me.
Checked `firstrun.sh` and this included the newline @jmagdev mentioned as well even on 1.8.5
I then flashed the image again and removed the newline, which allowed the Pi to boot and connect to the Wi-Fi
Removed I removed this newline and the Pi booted with Wi-Fi immediately.

I'm with @gsantner here, not happy with the changes to no longer allow the configuration to be done outside of Pi Imager.

GamerFrom61, 2025-06-14:
Going back to wpa_supplicant is not really an easy fix.

The team that develops Network Manager clearly state you should not create or modify the files directly but use nmcli for scripts as the layout could be changed or require other configuration files to be modified in the future.

Who knows what other info could need to be added as newer 802.11 standards are introduced?

Better to have a way of setting up the core detail in a Pi specific file rather than the old structure that needs to be left behind.

Jebble, 2025-06-14:
> Going back to wpa_supplicant is not really an easy fix.

> Better to have a way of setting up the core detail in a Pi specific file rather than the old structure that needs to be left behind.

Yeh it doesn't have to be the same file, but right now if your Pi doesn't connect to the Wi-Fi, the only option you have is to flash the image again, which isn't great.

gsantner, 2025-06-15:
> Try to change "rfkill unblock wifi" to "rfkill unblock all" in the firstrun.sh that should do the trick.

Thanks, I tried this. Does not work with 1.9.4. No device appearing in network after half hour.

b-morgan, 2025-06-15:
I had both 1.9.4 and 1.8.5 fail with the extra return on line 42 of `firstrun.sh`. Using 1.8.5, I deleted (`NO, CLEAR SETTINGS`) the customization settings and entered them again. I usually set the WiFi country code by typing "US" but this time, I scrolled through the (huge) list and selected "US" from the dropdown. The resulting `firstrun.sh` did NOT have the extra return.

imaerials, 2025-06-15:
Hi Guys! I faced the same issue . Got the firstrun.sh into vscode, analize it and found that there were some extra lines in coutry code and timezone.

Broken WiFi setup (lines 34-35)
Broken timezone setup 

From copilot found

1. **Line 34-35**: There's a syntax error in the `set_wlan` call. The country code 'AR' appears to be cut off and there's an incomplete line with just 'AR\n'.

2. **Line 54-55**: Similar issue with the `set_timezone` call where 'America/Argentina/Buenos_Aires' has a trailing newline character that's being treated as part of the string.

3. **Line 59**: The timezone assignment also has the same trailing newline issue.

## Specific Problems:

- **Broken WiFi setup** (lines 34-35): The country parameter is incomplete
- **Broken timezone setup** (lines 54-55, 59): Timezone strings have embedded newlines

## Fixed Issues:

✅ **WiFi Configuration**: Fixed the incomplete country code 'AR' parameter  
✅ **Timezone Configuration**: Removed embedded newlines from timezone strings  
✅ **Script Syntax**: All string parameters are now properly formatted

reruo321, 2025-06-15:
I solved such issue using my brand new Pi 5 and Imager v1.9.4 by removing newline typos in `firstrun.sh` from the end of these lines:
1. Inside the WiFi country code, starting with /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan
2. Inside the keyboard layout code, starting with /usr/lib/raspberrypi-sys-mods/imager_custom
3. Inside of the XKBLAYOUT string value

[image]

[image]

[image]

The lines to fix would be different on the different Pi versions or different OS settings. Make sure your editor uses Unix format EOL (LF), which can be done easily with Notepad++ in Windows.

I bought a new Pi 5 but failed so many times on imaging with custom WiFi setting for headless setup. I tried some experiments for three days, but no luck. On the first day, I reverted hostname, username, and WiFi country code (KR -> GB) to default, and toggled locale settings. On the second day, I change my WiFi router settings. On the third day, I downgraded the Imager to 1.8.5. Nothing worked. After I had connected the Pi to a display, I realized no WiFi setting was saved at all during many imaging trials. From here I found `firstrun.sh` typo would be the problem, so I removed all newline typos. After the first booting, my Pi now connects to WiFi very well.

b-morgan, 2025-06-18:
> What text editor were you using?

There is no "text editor" involved when entering customization settings in the RPi Imager. When checking the `firstrun.sh` for success or failure, I used Notepad++.

MontesDesigns, 2025-06-18:
> > What text editor were you using?
> 
> There is no "text editor" involved when entering customization settings in the RPi Imager. When checking the `firstrun.sh` for success or failure, I used Notepad++.

Got it. thanks. It seems this issue with the extra line on the WiFi code or Keyboard layout also breaks all other customizations set by the PPi Imager.

itinstructor, 2025-06-21:
> THIS WORKED!!!!!

I updated to Raspberry Pi Imager 1.9.4 on Windows 11. It worked fine for a while customizing the OS, then suddenly it stopped customizing. I used this solution, it works now. A very weird issue.

segadude, 2025-06-29:
I am having the same issue with a Raspberry Pi Zero W (Not Zero 2 W). I'm using Raspberry Pi Imager v1.9.4 on Windows 11 Home Version 24H2 OS build 26100.4351. I am using the [USBODE](https://github.com/danifunker/usbode) image, which is based on Bookworm. The Pi doesn't connect to Wi-Fi, which is important for this image because I need to connect to a web interface to use it. I tried flashing the image twice. The second time, I explicitly typed my Wi-Fi password instead of leaving the default in case it was wrong. The Wi-Fi still did not connect. I downgraded to Raspberry Pi Imager v1.8.5, and it successfully connected to Wi-Fi.

AyotoCorp, 2025-07-01:
I am having the same issue with a Raspberry Pi Zero 2 W, loading from a 64GB microSD card.  Using Raspberry Pi Imager v1.9.4 on Windows 11 Pro 24H2 OS build 26100.4351.  I attempted to install the Legacy 32-bit Lite OS (Bullseye), but none of the settings I attempted to adjust in the imager (hostname, username/password, SSID/password, localization settings, and SSH setting) were maintained when first booting the Pi.  Bizarrely, and contrary to other people's comments, downgrading to Imager v1.8.5 and v1.8.4 also didn't fix the issue.  When checking the firstrun.sh script, I can see my modifications are present.  I can also see the changes being applied according to my settings during initial startup if I use a monitor, but after a reboot the boxes to select keyboard layout and create a new user show up as if nothing happened.

lurch, 2025-07-01:
> Not a single time got Pi Zero W with the Pi Imager connected to network.

The common cause of that is people not waiting long enough for the Pi Zero to finish doing it's first-boot tasks (resize the root filesystem, generate SSH host keys, etc.), and so they try connecting to the Pi Zero before the SSH server is even running.
If you're able to (temporarily) connect an HDMI monitor and a USB keyboard to your Raspberry Pi Zero, that'll make it **much** easier to diagnose what might be going wrong. But that's a discussion that ought to be done at https://forums.raspberrypi.com/ rather than on GitHub.

gsantner, 2025-07-01:
It doesn't work when connected for hours at first boot either. Fresh written sdcard and keeo powered on over night - nothing in network.

Don't have a special mini-hdmi cable either around, which beside Pi Zero nothing else uses I'm aware off/have around.

DeeJay, 2025-07-02:
Imager v1.9.4 running on Windows 10 Home as of today, Weds 2nd July.

Image file 2025-05-13-raspios-bookworm-arm64-lite.img.xz

Customisation still introduces an unwanted (and damaging) New Line after Country Code.

```
if [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then
   /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan 'Coniston_Guest' 'b1a3a1ff72cbad57558161558163d5001085dbd21eea639c71fd9b1af5269591' 'GB
'
else
```

With that apparently extraneous byte the system does not proceed to make an association with the nominated hotspot.

Removing that byte 'manually' with a text editor before First Boot allows the wlan association to proceed normally.

bierbarrel, 2025-07-05:
Same here...pi4 4gb and wont save localization settings. Ill hook up to lan and run raspi config

ron02451, 2025-07-06:
Hello,
I had similar issues with a headless install.
Here is what I found.
using a hdmi cable , a monitor and a keyboard
run these commands
rfkill unblock wifi
nmcli radio wifi on
nmcli d wifi list
nmcli device wifi connect <AP name> password <password>
ifconfig -a

then make sure you get the ip address and make sure ssh is on through raspi-config

hamsjelv, 2025-07-09:
> Imager v1.9.4 running on Windows 10 Home as of today, Weds 2nd July.
>
> Image file 2025-05-13-raspios-bookworm-arm64-lite.img.xz
>
> Customisation still introduces an unwanted (and damaging) New Line [...]

The same happens with the keyboard keymap
`
if [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then
   /usr/lib/raspberrypi-sys-mods/imager_custom set_keymap 'dk
'
`

tugkan98, 2025-07-11:
> I have the same issue. Very much looking forward to seeing a patch soon as there is an open PR ([#1084](https://github.com/raspberrypi/rpi-imager/pull/1084))
> 
> Until the above PR is approved and shown as working, will offer steps for how I got it working on my end, as earlier instructions in this issue's comments didn't seem to always work for me:
> 
> 1. Use Raspberry Pi Imager v1.8.4
> 2. Clear OS customization settings and re-apply whichever you like, except for 'locale settings'. When selecting Wireless LAN Country, scroll down the list to select, no typing.
> 3. After flashing your media, open the bootfs drive/partition, and open the 'firstrun.sh' file with a text editor like Notepad++ (something that respects UNIX line endings). You may need to remove and re-insert the media drive. Don't use a RaspberryPi, as this file will be removed on first run & you'll have to re-flash the image.
> 4. Update the value on line 42 to include double-quotes. Save and safely remove the media drive for use with a RaspberryPi.
> 
> As an example- line 42 usually looks like this (for me): `country=US` Update to something like this: `country="US"`

This is the only solution worked for me!

Patel970, 2025-07-14:
Found a way to fix this:
Imager Version: v1.9.4
OS: Raspberry Pi OS Bookworm 64-Bit lite
Device: Raspberry Pi 4b
Storage Device: Sd Card
OS Customization: Yes, I was using OS customization when the bug occurred.

Way to fix it:

Step 1: Download the imager and flash the OS to your SD card. Then go through the initial setup process.
⦁		I got a message after logging in saying my wifi is being blocked by rfkill.

Step 2: Type in "rfkill list". It should look something like this:
0: hci0: Bluetooth
Soft blocked: no
Hard blocked: no
1: phy0: Wireless LAN
Soft blocked: yes
Hard blocked: no
. 
If this is true than run the command "sudo rfkill unblock" and "sudo rfkill unblock all" just to be sure then run "rfkill list". Then it should look like this
0: hci0: Bluetooth
Soft blocked: no
Hard blocked: no
1: phy0: Wireless LAN
Soft blocked: no
Hard blocked: no 
.

Step 3: Type in "sudo raspi-config" and navigate to System options and then wireless lan. Set that up and put your country and ssid and password in.

This all should fix the networking issue, but also I'm using ssh and with no customization it comes, by default, off. After this it is all optional:

Step 4(optional):
Type in "sudo raspi-config" and then navigate to interface options and then SSH and then enable ssh. This should fix it if you were going to use ssh.

Patel970, 2025-07-14:
Also I forgot this but you might also want to set the host name to do this type in "sudo raspi-config" and go to system opions and then hostname.

yordanovevgeni, 2025-12-30:
> Hello, I had similar issues with a headless install. Here is what I found. using a hdmi cable , a monitor and a keyboard run these commands rfkill unblock wifi nmcli radio wifi on nmcli d wifi list nmcli device wifi connect password ifconfig -a
> 
> then make sure you get the ip address and make sure ssh is on through raspi-config

This actually worked. At least it was able to unblock the wifi and after that used > sudo nmtui to connect.

RafaelKa, 2026-10-11:
Why it is closed? The issue still exists! If Imager has the settings applied, then WiFi can not be established at all.
The Imager writes something to the target, which leads to broken WiFi, even manual connecting fails then.

Please reopen the issue to fix it.
Who closed that issue and why?

Patel970, 2026-10-11:
This issue has been resolved in the latest version of the raspberry pi os imager