raspberrypi/rpi-imager #1071: [BUG]: Deb file has issues installing on Mint 21.3

MarkBinner, 2025-06-07:
### What happened?

I used imager version 1.8.5 and got a message there was an upgrade. I downloaded it and tried to install it on my Mint 21.3 system (based on Ubuntu 22.04 -which it claims is compatible). Instead of replacing the old version with the new one, which pretty much every deb file I've ever installed does, it gave a message saying there was a conflict with 1.8.5 version. I uninstalled 1.8.5 and tried to install again. This time I had a dependency error for util-linux. The version I have installed is too old 2.37 instead of 2.38. I reverted to 1.8.5 and now, when I use it, I get reminded there is a new version. The 1.9.4 AppImage appears to be OK but I deleted it having read about issues with wireless in other bug reports.

### Version

1.9.4 (Default)

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

Mint 21.3 MATE

### Selected OS

RaspAP

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

tdewey-rpi, 2025-06-09:
Thanks for the report, @MarkBinner 

I can confirm this looks like Raspberry Pi OS requirement that has been applied to the x86_64 build. Queued for 1.9.5.

tdewey-rpi, 2025-06-09:
Additionally, the package rename is intended. Previous builds relied on a artefact of how the Debian packaging system would handle the Architecture flag. The 1.9.x series uses a more explicit mechanism to defend against accidental cross-architecture releases.