raspberrypi/rpi-imager #1083: [BUG]: unable to write/paste in the textbox for ssh key

Mig-uana, 2025-06-15:
### What happened?

I'm unable to write or paste anything in the textbox where the public key supposed to be.

The other textboxes also act odd. There are delays/freezes when trying to type in the textboxes.
When I type in the text box nothing happens until "force a refresh" by switching to an other tab and go back this doesn't work for the ssh public key textbox.

I made a quick and dirty screen recording to make it clear

https://github.com/user-attachments/assets/810be744-1863-4121-81a0-5d484b397b23

 

### Version

1.9.4 (Default)

### What host operating system were you using?

Windows

### Host OS Version

10.0.19045 Build 19045

### Selected OS

raspberry pi os bookworm 32 bit

### Which Raspberry Pi Device are you using?

Raspberry Pi A, B, A+, B+, and Compute Module 1

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

Mig-uana, 2025-06-16:
Hello, @tdewey-rpi yes, using tab to switch to the next control also updates the textbox.

I can't tell what the issue is with the SSH key field I'm trying to paste the private they which is a long string of characters. 
After pasting in to that textbox the textbox seem to lose it's right click functionality for a few seconds.

Mig-uana, 2025-06-16:
@lurch I'm using a nvidia 4070 S GPU and I updated the divers but no luck.
I tried to use the previous version of rpi imager (1.9.0) and that works fine so I don't think my PC is the problem, but on the hand the textboxes worked fine when I ran the Imager in windows sandbox.

tarasmurzenkovv, 2025-06-29:
Same happening to me using the latest version of the imager

tdewey-rpi, 2025-07-21:
1.9.6 also includes more helpful error messages. Closing as fixed.