raspberrypi/rpi-imager #1085: [BUG]: ssh pubkey: paste rejected without feedback ("+" or even "." as part of a foo+bar@example.com comment)

cloph, 2025-06-16:
### What happened?

Trying to customize the installation and setting login credentials by ssh is awful experience, since while copy'n'paste seems to work in other fields (so one can paste the wifi password without problem), setting up ssh-keys seemingly breaks for no reason at all. Typing works, but who wants to type a more or less random string only to later find out logging in won't work because of a missing character...

issue #1083 has an important hint: 

> the SSH key field has input validation that may not like the contents of your clipboard.

With that bit of info I figured out why it wasn't accepting my ssh key. I was trying to copy it with the comment part, and that typically is an email address. I use a + from my gmail ones for different keys, but even removing that still won't work since it also rejects entering a . so no email addresses are allowed. It accepts the @ so user@host still would work.

tldr:
**There should be visible feedback when the paste is rejected** because of "invalid" characters, or paste should be accepted and a warning added about potentially being malformed. Or paste until the problematic character and strip the rest
And of course email addresses should be accepted as part of a key-comment/identifier/the validation is way too strict.
also related: issue #629

### Version

1.9.4 (Default)

### What host operating system were you using?

macOS

### Host OS Version

15.5

### Selected OS

PiOS lite

### Which Raspberry Pi Device are you using?

Raspberry Pi A, B, A+, B+, and Compute Module 1

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
no log/feedback and that is the biggest problem.
```

lurch, 2025-06-16:
Yeah, sounds like the comment-part of the regex at https://github.com/raspberrypi/rpi-imager/blob/qml/src/OptionsServicesTab.qml#L109 needs to be updated to be more lenient.

zimmerman-pliant, 2025-06-20:
Yes, this is a serious serious inconvenience.

I'm having this issue on Windows with 1.9.4.

I can no longer paste multiple keys to set up shared resources based on a file of team pubkeys, and I can no longer paste a single `ecdsa-sha2-nistp521` key. The keys I want to enter are a mix of types.

Only keys starting with `ssh-` are accepted, but `ssh-keygen` doesn't generate that for `ecdsa` keys:

```
ssh-keygen -t ecdsa -C throwaway-user@computer
```

generates a key like:

```
ecdsa-sha2-nistp256 AAAAE2VjZHNhLXNoYTItbmlzdHAyNTYAAAAIbmlzdHAyNTYAAABBBPBvDqW3PuyGlCjMUIx5GO+d7aE9jce4lafvfk06emZwPHdXsyW4FV7EtJwnCKSnIA4yciUIlEQpOEct0SkX4PA= throwaway-user@computer
```

This is also not accepted. Only keys that start with `ssh-` are accepted (playing around with manual typing into the field here):

[image]

As a workaround for now I'm going to make an `ed25519` key to log in and then I can add the rest later, but this is inconvenient.

lurch, 2025-06-24:
Collating other related issues into this one:
* #1095 points out that the current regex also disallows key-encodings that contain a `+` - this appears to be required due to [different base64 representations using different encoding alphabets](https://stackoverflow.com/questions/22108263/plus-sign-in-base64-encode64-output)
* #1092 points out that the current regex also disallows "security token stored keys" like `sk-ssh-ed25519@openssh.com`

JacquesOfAllTrades, 2025-07-09:
> In the meantime, note that the email-address in your public-key is just a "comment", so you can (temporarily) change it in the RPi Imager settings (to bypass the validation bugs in v1.9.4), with no ill effects.

There's still the issue that the imager rejects key types that don't start with `ssh`. For example, I have a user with an `ecdsa-sha2-nistp256` key who essentially can't use the current imager.

tdewey-rpi, 2025-07-18:
1.9.5 is now available in the releases area if you have this problem, and 1.9.6 is incoming, which will be the next prompted upgrade.