raspberrypi/rpi-imager #1086: [BUG]: UI update on second image flash is buggy

benoit74, 2025-06-16:
### What happened?

Steps to reproduce:
- select a custom image
- select any storage (was an externally attached NVMe disk on a USB-C reader in my case, did not tried other media)
- flash the image
- wait for completion
- eject storage (and source image which was also stored externally, don't know if it makes any difference)
- insert new storage (and insert source image again)
- flash the image again

Here the status update stays at "Verifying 0%" (while it is writing in fact). The button stays at "Cancel verify". Cancelling the verify does not work. Quitting the application (via the red cross from "apple controls") causes the application to crash (after asking for confirmation since a pending task is ongoing). See details attached

This is a recent regression, worked well in 1.9.0

[rpi-imager-crash.txt](https://github.com/user-attachments/files/20757349/rpi-imager-crash.txt)

I'm not sure this is only a UI issue or a more significant one (don't have time to debug further, sorry about that)

### Version

1.9.4 (Default)

### What host operating system were you using?

macOS

### Host OS Version

Sequoia 15.5

### Selected OS

Custom

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

NVMe SSD via Third-party PCIe attachment

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-06-16:
Known problem (e.g. #1073 ), which should be fixed in the upcoming 1.9.5

tdewey-rpi, 2025-07-21:
Fixed in 1.9.5+

closed 2025-07-21 by tdewey-rpi