raspberrypi/rpi-imager #1092: [BUG]: Unable to add ssh public keys of type sk-ssh-ed25519@openssh.com & keys with comment to OS customisation

beachrunner2k, 2025-06-21:
### What happened?

I've just upgraded to Raspberry Pi Imager v1.9.4 and noticed the OS customisation page for adding ssh public keys has changed. Previously it was effectively a free text field that appeared to be used to populate ~/.ssh/authorized_keys. With v1.9.4 there are now discreet entries for each ssh key. I am experiencing the following issues:

- Cannot copy/paste ssh public keys of type sk-ssh-ed25519@openssh.com (security token stored keys, e.g. Yubikey) in the text field. Example key below.
`sk-ssh-ed25519@openssh.com AAAAC3NzaC1lZDI1NTE5AAAAINgHoMCWf+B7AVfIP58SSRIvENXVKx3Vq/Luj8/6rvLA`

- Cannot copy/paste ssh public keys containing a comment. Example below. It is however possible to manually enter the comment after the key string has been pasted.
`ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAINgHoMCWf+B7AVfIP58SSRIvENXVKx3Vq/Luj8/6rvLA email@example.com`

### Version

1.9.4 (Default)

### What host operating system were you using?

macOS

### Host OS Version

Sequoia

### Selected OS

Raspbery Pi OS 64-bit Bookworm

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-06-21:
Related to #1085

tdewey-rpi, 2025-07-21:
Fixed with 1.9.5+

closed 2025-07-21 by tdewey-rpi