raspberrypi/rpi-imager #1094: [BUG]: Raspberry Pi Imager incorrect syntax in firstrun.sh causing wifi issues (Raspberry Pi OS Lite x64)

theolensky, 2025-06-22:
### What happened?

Hello,

There's an issue with the imager on the lastest version (1.9.4) where line breaks appear in firstrun.sh causing the script to not work properly and therefore not connect to wifi. 

I want to give full credits to @sakuv3 who kindly pointed this out in #901  and has saved me many hours....

"After inspecting the generated firstrun.sh, I noticed that some parameters contained unexpected newline characters (\n) — specifically:

    XKBLAYOUT="de\n" instead of XKBLAYOUT="de"
    set_wlan '[SSID]' '[PSK]' 'DE\n' instead of 'DE'
    /usr/lib/raspberrypi-sys-mods/imager_custom set_keymap 'de\n' instead of 'de'

These newlines broke both the keyboard layout config and the WiFi setup silently."

I can confirm the same is still happening and I can reproduce on my Pi. 

Please could the team update the generation of the firstrun.sh to remove the 3 unnecesary new line breaks?

### Version

1.9.4 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Windows 11 23H2

### Selected OS

Raspberry Pi Os Lite (64-BIT)

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

sakuv3, 2025-06-22:
#1060

lurch, 2025-06-23:
Looks like this might also be fixed by https://github.com/raspberrypi/rpi-imager/pull/1084/files#diff-8625d58f57ae280809d073a633e766757ea0f518788ce2b7d7eceb854f3d7ee9R360-R364 in #1084 ?

RobertWojtowicz, 2025-07-04:
> Same issue on pi zero w, imaging raspbian lite 32bit. Device doesn't join the network. I applied my own firstrun.sh as below, worked immediately.
> 
> ```
> #!/bin/bash
> set +e
> systemctl enable ssh
> echo "pi:raspberry" | chpasswd
> /usr/lib/raspberrypi-sys-mods/imager_custom \
>     set_wlan 'my_wifi_ssid' 'my_wifi_password' 'GB'
> rm -f /boot/firstrun.sh
> sed -i 's| systemd.run.*||g' /boot/cmdline.txt
> exit 0
> ```

confirm works :)

BR,
Robert

CraysonM, 2025-07-05:
When the imager is run on a linux machine, neither of these problems occur. It seems the firstrun.sh is properly formatted there.

numbworks, 2025-07-09:
I have the same issue!

- **HOST OS**: `Ubuntu 25.04`
- **APP**: `Raspberry Pi Imager 1.9.4`
- **TARGET DEVICE**: `Raspberry Pi 3`
- **TARGET IMAGE**: `Raspberry Pi OS Lite (64 bit) - May 13th 2025`

Before flashing, I add host name, wifi, etc via "Edit Configurations".
The image gets flashed to the microSD and boots on the  Raspberry Pi just fine, but without connecting to the wifi.

If I do exactly the same thing using `Ubuntu 24.2.x LTS` as target image, everything works as expected.

lurch, 2025-07-19:
Please give v1.9.5 a try, as it should fix this problem https://github.com/raspberrypi/rpi-imager/releases

tdewey-rpi, 2025-07-21:
Fixed with 1.9.5.

closed 2025-07-21 by tdewey-rpi