raspberrypi/rpi-imager #1095: [BUG]: Unable to paste a new SSH key

impresionista, 2025-06-23:
### What happened?

When I go into a OS Customization dialog and try to add a new SSH key, the interface won't paste anything from my clipboard. I've also noticed I can't write anything, so I suspect this error has something to do with the regex validation on the string.

Here is the public key I've tried:
`ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIFy1SqMQRxoV53nvYc0WCQGmO0XHmXdOsAliIIooRC/+ marcos@felipe`

I've tried adding the key manually after install and it worked without issues.

If it's in fact regex not valid, it would be useful to show an error/warning message.

### Version

1.9.4 (Default)

### What host operating system were you using?

Other Linux environment

### Host OS Version

Arch Linux 

### Selected OS

Raspberry PI OS Lite (64bits)

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-06-23:
Sounds like a duplicate of #1085

impresionista, 2025-06-23:
Sorry for the extra report, I've missed the existing ones. Feel free to close this issue if you need to. I'll try to add all the feedback I can in #1085 to avoid spreading relevant information in multiple issues.

tdewey-rpi, 2025-06-24:
Thank you, @impresionista 

Closing as a duplicate, will continue in #1085