raspberrypi/rpi-imager #1105: [FEATURE]: Debug builds

SkyLined, 2025-07-08:
### Is your feature request related to a problem? Please describe.

I believe I have found reliable memory corruption in Raspberry Pi imager. It is hard to debug on Windows without debug symbols. There does not appear to be a debug build, or symbol server available. Would it be possible to create a debug build, or make symbols available?

### Describe the solution you would like to see implemented

Add debug builds with symbols to the release, or make the symbols available through a symbol server.

### Describe alternatives you've considered

Debugging an application I know nothing about without symbols is too time consuming. Building my own version is also too time consuming.

### Additional context

I expect anyone not familiar with the project who encounters a crash would benefit from debug builds.

### Version

1.9.4 (Default)

lurch, 2025-07-08:
_I expect anyone not familiar with the project who encounters a crash would..._ create a new GitHub issue describing the problem in detail, rather than trying to debug the problem themselves? 😉 

Although I admire your ambition! ❤️  I think we assume that anyone that's prepared to debug the problem themselves, would also be prepared to build their own version of the program from the sourcecode; as that would also allow them to test any potential fixes? 🤔

tdewey-rpi, 2025-07-08:
The memory corruption issue has been resolved in the upcoming 1.9.5, and was previously covered in #965.

Concerning publishing debug symbols - I'm not convinced this represents the best path forward here. Setting up external symbols is not something I would consider easier than following the build instructions included in the README - which I've put a non-trivial amount of effort into simplifying over the last year or so.

As such, I'm considering rejecting this feature request - unless I can see demonstrable evidence that publishing an additional build type is easier/more complete than guiding people to reproduce the build.

closed 2025-07-21 by tdewey-rpi