raspberrypi/rpi-imager #1107: [BUG]: prompt window popped up saying 'partitions does not exit'

cjqwjbdx, 2025-07-10:
### What happened?

A bug happened!The SD card used normally before could not be burned, and a prompt window popped up saying 'partitions does not exit'

### Version

1.9.4 (Default)

### What host operating system were you using?

Windows

### Host OS Version

22H2

### Selected OS

Raspberrypi OS x64(lastest)

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
partitions does not exit
```

lurch, 2025-07-10:
Are you able to use the "Operating System" -> "Erase" option to restore your microSD card to a working state?

cjqwjbdx, 2025-07-10:
I used the Windows diskpart command to restore my sd card into the working state.I tried amost all the sd card in my home.And the bug is still there.But when I use Ver1.8.5,the Bug disappear.

已发送: 2025 年 7 月 10 日 星期四 18:58
收件人: raspberrypi/rpi-imager ***@***.***>
抄送: cjqwjbdx ***@***.***>; Author ***@***.***>

cjqwjbdx, 2025-07-10:
I'm very glad that you give my a reply.Thank you!