raspberrypi/rpi-imager #1111: [BUG]: Very slow and freezing UI

Zoonie81, 2025-07-11:
### What happened?

HI!

Updated to 1.9.4, which is the latest exe on rpi website and the imager became unusable. It starts up but **any** button press or choice in **any** of the 3 lists is delayed by 30-60 secs. Scrolling through the list of options in Operating System also makes the UI freeze for a long period of time. This is the first time I experience this with the imager and I've been using it since the early version.

I tried uninstalling and reinstalling 1.9.5 but no go. Reboot Windows ofc.

PC running:
Intel 5960X
128Gb RAM
Asus GeForce 3090
Sony MRW-G2 card reader

Are there any links for older versions of the imager I can try?

### Version

1.9.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Windows 10 22H2

### Selected OS

HAOS

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-07-12:
I think that one of the significant changes between the 1.8.x releases and the 1.9.x releases was the switch from Qt 5 to Qt 6. So could you check that your graphics drivers are up to date please, just in case there's some kind of weird compatibility quirk?

Zoonie81, 2025-07-12:
@lurch thanks for the reply, and sorry for being short in my description. I did update to latest Nvidia drivers last night in my search for a solution, but it didn't help. I also tried a full uninstall/reinstall. As soon as I went back to 1.8.5 all problems vanished. Let me know if I can help figuring this out with more tests/provide data. 

Regards, Z

Zoonie81, 2025-07-12:
Hi @tdewey-rpi 

Just checked. Forcing software resolves the issue. FYI, I'm on nvidia 576.88 drivers at the moment. Not sure what I had before updating last night, but at least a few months older drivers. The card is an ASUS ROG 3090 OC. Hope this helps.

Regards, Z

lurch, 2025-07-12:
I did a bit of searching and found https://doc.qt.io/qt-6/windows-graphics.html which suggests that you can force Qt to use Direct3D 12 instead of 11, but as I don't use Windows myself I don't know whether that'll help or not?
(And of course if software rendering works fine, you could just stick to that workaround)

Zoonie81, 2025-07-12:
Hi @tdewey-rpi 

This is a very capable machine, used for 4K video editing and many other graphic heavy workloads, like CAD/3D work etc. I have checked GPU load while the Imager is open and the utilization is around 1-3%. After reading your post I also ran D3D 11 and D2D 11 tests with PC Mark and 3D Mark. No issues and very high results. I've also checked several programs that run in the background and everything is up to date and functioning correctly. Not saying you're wrong, but highly unlikely this is a local D3D problem. I could be wrong of course.

@lurch, thanks for the suggestion. Seeing the CPU can handle software rendering just fine, I'll keep using it this way for now not to mess with possible other software relying on Qt.

Zoonie81, 2025-07-15:
@tdewey-rpi I hope I didn't come off as arrogant. My comment was in reply to your theory about there being another D3D program consuming GPU time and D3D 11 misbehaving on this system. 

I do however agree with you that this probably is related to Qt6. In any case, problem "solved" with software render. Thanks for the help.

closed 2025-08-01 by tdewey-rpi

geerlingguy, 2025-08-04:
I've just updated to 1.9.6, and today some of the UI on my Mac (M4 Max Mac Studio, rebooted to be sure nothing else was funky) is stuck frozen for a looooong time. Like just clicking 'Yes' on the 'Would you like to apply OS customization settings?' takes minutes, and I've had to force quit a bit.

Not sure what's going on, but I've flashed about 15 microSD cards and USB drives today, and it started getting really sluggish in the past hour.

geerlingguy, 2025-08-04:
I also downloaded 1.8.5 just to use the older UI, and it was having the same trouble—extreme slowdown with the process following the 'would you like to apply changes' portion...

lurch, 2025-08-04:
> I also downloaded 1.8.5 just to see if the older UI made a difference, and it was having the same trouble—extreme slowdown with the process following the 'would you like to apply changes' portion...

I just noticed [this issue](https://github.com/balena-io/etcher/issues/4505) over on the Balena Etcher repo that is also claiming that older "previously good" versions of Etcher are now freezing on macOS. Maybe there's been a change in the macOS drivers which has borked something? 🤔

Kelketek, 2025-08-22:
Just wanted to bump this as the problem is occuring for me on Linux as of 1.9.6. Saving customization dialog gets stuck freezing for a laboriously long time-- like minutes. Same for 'Yes' afterward. It only started after I'd done several imaging tasks over the course of the day. Then at some point it just got super slow after being fine for a long time.

Rebooting didn't fix it. `apt purge rpi-imager-amd64` and then reinstalling the deb didn't fix it either.

Kelketek, 2025-08-22:
Update: I have discovered that `~/.config/Raspberry Pi/Imager.conf` has ballooned to 12MB in size. Most of it is my SSH key repeated over and over again. There must be something causing the key to be repeatedly be copied, and thus it is likely related to #1168.

Workaround, for anyone finding this, is to remove the Imager.conf and start over with your custom settings.

geerlingguy, 2025-08-25:
After reading @Kelketek's post about the key ballooning in size, I realized after re-upgrading to 1.9.6 today, I tried deleting my SSH pubkey, then it re-populated again. Tried again, it came back... and I'm wondering if maybe that could be a culprit here for me.

I always write my key to the image, though I haven't edited the key in a couple years. But I can't find a file inside `~/.config/Raspberry Pi` (nor even that directory), so I wonder if it's stored somewhere else on the Mac?

jedahan, 2025-08-25:
Mine was in `~/Library/Preferences/org.raspberrypi.Imager.plist`. I do not think I have this issue, but I also haven't been using RPI imager lately, just in case people wanted another pointer.

Biorix, 2026-08-31:
@Kelketek 
I'm on V1.9.6 (snap) I and still have the same issue after repeated image creation 
my file ~/snap/rpi-imager/current/.config/Raspberry Pi/Imager.conf is full of repeated ssh key

PR #1156 (2025-08-01): 1.9.6: Fix roll-up
- Added conditional ejecting of the disk based on the _ejectEnabled flag in DownloadExtractThread.
- Implemented saving of miscellaneous options (beep, eject, telemetry) to persistent storage in OptionsPopup.

Fixes #1154 
Fixes #1089
Fixes #1148 
Fixes #1152 
Fixes #995 
Fixes #1111