raspberrypi/rpi-imager #1115: [BUG]: Wifi and ssh options don't work and boot partition called bootfs

edrickwong, 2025-07-12:
### What happened?

Hi there,

I just downloaded the Raspberry Pi Imager for Windows today. It seems to reformat the image as bootfs (instead of boot). Also, even with the Wifi and ssh options selected, my pi zero 2w doesn't seem to connect. I took a look into the contents, and no ssh or wpa_supplicant.conf files were created.

### Version

1.9.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

23H2

### Selected OS

Raspberry Pi OS 64-Bit 2025-05-13

### Which Raspberry Pi Device are you using?

Raspberry Pi Zero 2 W

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-07-13:
> It seems to reformat the image as bootfs (instead of boot).

I don't know what you mean by this - could you explain it a bit more please? Note that RPi Imager "just" copies the OS image directly onto the SD card, so partition names, partition sizes, etc. are entirely determined by what's in the raw disk image.

> I took a look into the contents, and no ssh or wpa_supplicant.conf files were created.

Correct. These files were the "old" way of configuring Raspberry Pi OS. These days, RPi Imager configures an OS by writing a `firstrun.sh` file to the boot partition.

Gaatvul, 2025-07-14:
Can confirm that this is an issue for me as well. 
When looking in `firstrun.sh`, it appears that the passkey for WIFI is being encoded to hex, but perhaps is not being decoded back to plaintext.
When I manually edited the passkey strings back to decoded plaintext, the pi connected immediately, was visible on the network, and I could SSH into it without any issue.

[image]

### **Version**
1.9.4

### **What host operating system were you using?**
Windows 11 Home

### **Host OS Version**
23H2

### **Selected OS**
Raspberry Pi OS Lite (64-Bit)

### **Which Raspberry Pi Device are you using?**
Raspberry Pi 3 A+

### **What kind of storage device are you using?**
microSD Card in an external reader

### **OS Customisation**

- [x] Yes, I was using OS Customisation when the bug occurred.

edrickwong, 2025-07-19:
For connecting to Wifi, it seems like the first SD card I imaged worked, but then using the same profile, imaging a second SD card didn't work. When I connected the second one to the screen, I was prompted with the Wifi Authentication popup, and the pre-filled password still seemed like an encoded string.

twoscomp, 2025-07-19:
Updating to 1.9.5 had fixed the customization issues for me. Now booting with Wi-Fi successfully.

lurch, 2025-07-19:
> I was prompted with the Wifi Authentication popup, and the pre-filled password still seemed like an encoded string.

Yes, it stores a hashed version of your Wifi password, so that (for security/privacy reasons) it isn't stored in plain-text on the 'boot' partition of your SD card.

> For connecting to Wifi, it seems like the first SD card I imaged worked, but then using the same profile, imaging a second SD card didn't work.

So if it worked the first time, but didn't work the second time, that suggests that you must have typed in the password incorrectly when you were customising the second card? 🤔

tdewey-rpi, 2025-07-21:
Yes - Imager will perform a one-way hash of the WiFi password, as the networking stack in Raspberry Pi OS allows. This is explicitly designed to mitigate the 'over the shoulder' password compromise.

Further, in the 1.9.5+ releases, 'show password' functions have been removed, to avoid the confusion this creates.

tdewey-rpi, 2025-07-31:
Closing as no further reports following 1.9.5+ release.

closed 2025-07-31 by tdewey-rpi