raspberrypi/rpi-imager #1116: [FEATURE]: Add image support under General OS section for Tiny Core Linux

computeronix, 2025-07-15:
### Is your feature request related to a problem? Please describe.

The image does not exist today and as a result you manually go to the site - http://tinycorelinux.net/ - and identify the proper build based on the Raspberry PI being used, which is actually pretty buried, and then select and download and custom upload it, then it works.

### Describe the solution you would like to see implemented

rpi-imager should offer in the General OS section Tiney Core Linux as an option and have it provide the relevant versions and/or options of the Tiny Core Linux download versions that are applicable based on the Raspberry PI model selected based on the capabilities of the Raspberry PI. For example if you have a Raspberry PI 2 you can only use this the builds in the armhf area : https://distro.ibiblio.org/tinycorelinux/16.x/armhf/ however newer Raspberry Pis can support different architectures as a result and they are listed here: http://tinycorelinux.net/ports.html

### Describe alternatives you've considered

Manual download as described above then using the custom image to accept and write to it which works

### Additional context

Tiny Core Linux Site: http://tinycorelinux.net/

Tiny Core Linux Distros: (as of this writing 16.x is latest) - https://distro.ibiblio.org/tinycorelinux/16.x/

### Version

1.9.5 (Default)

lurch, 2025-07-16:
This is something that the TinyCoreLinux developers would need to add themselves - they'll need to host a subitems JSON file on their site, and then fill out an "application form". See e.g. #1108 and #1025 for more info.
So if this is a feature you'd like to see in Raspberry Pi Imager, you'll have to ask the TinyCoreLinux team...

tdewey-rpi, 2025-07-16:
Closing as 'Won't address' - @lurch has already cited why.