raspberrypi/rpi-imager #1123: [BUG]: I can't connect to wi-fi and next connect by ssh

mat-jan, 2025-07-19:
### What happened?

A bug happened!

I flash my SD card for RaspberryPi Zero WH and i can't connect to my RPi. I tried 2 times and from two place and it didn't worked, so I report that to you.

### Version

1.9.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

23H2

### Selected OS

RaspberryPi OS Full 32bit

### Which Raspberry Pi Device are you using?

Raspberry Pi Zero, Zero W, Zero WH

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

michaelbapst, 2025-07-20:
I had the same issue, for me it turned out that I hadn't set the WIFI Country setting in the Customization settings.

tdewey-rpi, 2025-07-21:
Added Wi-Fi Country validation to the OS customisation options in 1.9.6.

closed 2025-12-12 by tdewey-rpi