raspberrypi/rpi-imager #1125: [BUG]: Username validation prevents usernames starting with numbers on MacOs (regression from Windows version)

0xneobyte, 2025-07-20:
### What happened?

## Bug Description
The macOS version of Pi Imager prevents entering usernames that start with numbers (e.g., `0xneobyte`), while the Windows version allows the same username without issues.

## Steps to Reproduce
1. Open Pi Imager on macOS
2. Go to OS customization/advanced settings
3. Try to enter a username starting with a number (e.g., `0xneobyte`)
4. The number characters are not accepted/typed

## Expected Behavior
Should be able to enter usernames starting with numbers, consistent with the Windows version behavior.

## Actual Behavior
Cannot type numbers at the beginning of the username field on macOS.

## Environment
- **Pi Imager Version:**  v1.9.4
- **Operating System:** macOS 15.5 (24F74)
- **Previously working on:** Windows with same username format

## Additional Context
This appears to be a platform-specific validation issue. The same username (`0xneobyte`) works fine in the Windows version but is blocked in the macOS version. This creates inconsistency between platforms and breaks existing workflows when switching between operating systems.

## Workaround
Currently using alternative username formats or renaming after first boot with `usermod -l`.

### Version

1.9.4

### What host operating system were you using?

macOS

### Host OS Version

15.5 (24F74)

### Selected OS

Raspberry Pi OS

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-07-20:
Yeah, looks like it [validates](https://github.com/raspberrypi/rpi-imager/blob/qml/src/OptionsGeneralTab.qml#L190) the username using the regex `^[a-z][a-z0-9-]{0,30}$` (i.e. it has to start with a lowercase letter).

0xneobyte, 2025-07-21:
Then it's only for macOS imager ? 😭 just checked the windows version and it's the same version 1.9.4 and I used 0xneobyte as username

tdewey-rpi, 2025-07-21:
Confirmed on 1.9.6, no observable platform difference.

Closing this one as 'Won't fix', because relaxing the restriction would appear to cause people subtle & interesting problems later on, depending on what packages they choose to install or use.

closed 2025-07-21 by tdewey-rpi