raspberrypi/rpi-imager #1152: [BUG]: Duplicate SSH keys added to authorized_keys

HanSooloo, 2025-07-30:
### What happened?

I have one SSH key entered into the customized settings.  However, after running the imager, and setting up the RPi, I end up with duplicate lines in `authorized_keys`.
Inspecting `/boot/firstrun.sh`, I can see duplicated parameters supplied to `imager_custom enable_ssh`

### Version

1.9.6 (Default)

### What host operating system were you using?

macOS

### Host OS Version

15.5

### Selected OS

Raspberrypi OS 64 bit Lite Bookworm

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
user@pi-hole-02:~ $ cat .ssh/authorized_keys
ssh-ed25519 <redacted> user@local
ssh-ed25519 <redacted> user@local
```

HanSooloo, 2025-07-31:
It's funny that I opened an issue of "duplicate entries" and created a duplicate => #1153  .. was waiting to submit the issue, accidentally hit Back, lost all text and retyped everything and click Create .. I guess browser submitted 2 different issues.  Sorry about the dupe.

Bra1nsen, 2025-08-08:
Why isnt it already released??

PR #1156 (2025-08-01): 1.9.6: Fix roll-up
- Added conditional ejecting of the disk based on the _ejectEnabled flag in DownloadExtractThread.
- Implemented saving of miscellaneous options (beep, eject, telemetry) to persistent storage in OptionsPopup.

Fixes #1154 
Fixes #1089
Fixes #1148 
Fixes #1152 
Fixes #995 
Fixes #1111