raspberrypi/rpi-imager #1168: [BUG]: SSH login does not work after writing image for Raspberry Pi 1B+ in version 1.9.6, but not in 1.9.0.

sumver, 2025-08-16:
### What happened?

1.9.6 Raspberry Pi Imager after writing pre-downloaded system image for Raspberry Pi 1B+, no matter how I set the password, I always get a message that it's not correct 
Here's my Raspberry Pi Imager setup: 
username:sumver 
password:12345678 
enable SSH:yes
use password authentication:yes 
Otherwise, keep the defaults.

When I log in with ssh, I get the following message: SSHAuthFailError(All authentication methods failed)

I've tried to test the burn image several times with version 1.9.0, using the exact same settings, and SSH logs in correctly!

I've tested this with two Raspberry Pi 1B+, on both version 1.9.6 and 1.9.0, and I've concluded that there is a bug in version 1.9.6 that prevents SSH login.

### Version

1.9.6 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Win11 Version24H2

### Selected OS

Raspberry Pi OS (Legacy) Lite-6 May 2025- Debian 11 

### Which Raspberry Pi Device are you using?

Raspberry Pi A, B, A+, B+, and Compute Module 1

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

sumkincpp, 2025-09-06:
Can confirm - ssh password auth doesn't work with 1.9.6, 
works pretty well on 1.9.4 

probably the bug was introduced here https://github.com/raspberrypi/rpi-imager/pull/1113/files

danielwdunn, 2025-10-08:
I was able to reproduce this issue for a Raspberry Pi Zero 2 W.

Heinrich4771, 2025-11-30:
Confirm, I've the same issue with a raspberry pi5. 
Used the 2.0 Imager on a Mac to install the latest Raspberry PI OS (64bit) yesterday. 
Set SSH to on and set authentification to password. Disabled PiConnect.
Wrote the image to the sd card. Pi5 booted from the sd card. The Mac Terminal sees the PI but connecting via SSH was refused by the PI5.
 
Regards
Heinrich

TheDireHavok, 2026-05-01:
I have this same issue with Bookworm