raspberrypi/rpi-imager #1178: [BUG]: This program does not support the version of Windows your computer is running.

InterLinked1, 2025-08-26:
### What happened?

After the first page of the install wizard which prompts for language selection, the installer errors out:

[image]

### Version

1.9.6 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Windows 7 SP1, x64

### Selected OS

N/A

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

Other

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

InterLinked1, 2025-08-26:
1.9.4 and 1.9.6 are both broken like this. 1.8.5 works properly.

lurch, 2025-08-26:
It's a bit hidden away, but see the "System Requirements" at https://github.com/raspberrypi/rpi-imager/releases/tag/v1.9.4

tdewey-rpi, 2025-08-26:
> Are you saying the installer intentionally blocks old versions? I can't see a good reason to do that preemptively. The majority of software still works just fine on Windows 7 with no special modifications or considerations. This software isn't complex enough either really to warrant not retaining backwards compatibility.
> 
> In any case, if you are intentionally blocking operating systems, the README and change log should list a minimum Windows version requirement.

As I've stated elsewhere:

If the OS vendor is no longer willing to stand by that version, then continuing to support it simply means an ever-widening support cone for the maintainers - and a potentially ever-growing amount of time being spent on an OS that will have an ever-decreasing number of users.

Further, the application depends on Qt, which requires at least Windows 10: https://doc.qt.io/qt-6/windows.html

This restriction is communicated through the releases page, and by the installer - preventing you from installing something that has no expectation of working.

Closing as 'By Design' - I don't have the bandwidth to support an older Qt version, including all the various bugfixes that would need to be backported. For reference, Windows 10 was released over 10 years ago at this point (July 2015).

closed 2025-08-26 by tdewey-rpi