raspberrypi/rpi-imager #1179: [BUG]: Telemtry setting not remembered between imaging multiple cards

0xDE57, 2025-08-26:
### What happened?

'Enable Telemetry' flag is always reset to **on** between imaging cards.

When the prompt "Would you like to apply OS customization settings?" appears. In Edit Settings 
OS Customization > Options > Enable Telemetry

I turned Telemetry **off**. Imaged a card. Inserted another one, and double checked the settings and found the flag was reset back to enabled. I then imaged another card and it again was enabled.

It remembers every other setting like username, password, time zone, public-key, but seems not to remember the state of telemetry flag.

I did find the `--disable-telemetry` and will be using that from now on. 

>Users beware: turn off the telemetry flag between cards...

### Version

1.9.6 (Default)

### What host operating system were you using?

Other Linux environment

### Host OS Version

n/a

### Selected OS

n/a

### Which Raspberry Pi Device are you using?

Raspberry Pi 5, 500, and Compute Modules 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

whc2001, 2025-09-17:
Might be designed behaviour, also all the settings on `Options` page is not saving for me

0xDE57, 2025-09-18:
If it is intentional, then this is functionally a dark pattern. 

I explicitly turned off telemetry. That doesn't mean turn it back on when I am not looking...

whc2001, 2025-10-01:
> > If it is intentional, then this is functionally a dark pattern.
> > I explicitly turned off telemetry. That doesn't mean turn it back on when I am not looking...
> > Ideally it would be opt-in, not opt-out anyways.
> 
> This was not intentional - it was a bug in settings persistence that affected an entire category of settings.
> 
> Please check v2.0.0-rc1, and confirm that behaves in the manner you expect.

Can confirm it works fine, also the newer config UI flow feels better, thanks!

[image]

closed 2025-11-05 by tdewey-rpi