raspberrypi/rpi-imager #1185: [BUG]: Security Issue | OS Customisation is re-adding ~\.ssh\id_rsa to ~/.ssh/authorized_keys even when DELETE KEY button used.

liamriles, 2025-08-29:
### What happened?

## Steps to Reproduce:

1. Have an id_rsa.pub in  ~\.ssh\ on Windows
2. Open RPI Imager OS Customisation
3. Enable SSH → "Allow public-key authentication only"
4. Add your custom public key
5. Click "DELETE KEY" to remove the other auto-populated key
7. Ensure there are no other keys (use scroll wheel while hovering over key rows to check none hiding with UI glitch)
8. Click "Save"
9. Image the SD card
10. Boot Pi, SSH in and run: cat ~/.ssh/authorized_keys

## Expected Behavior:
Only the custom key should be present in authorized_keys

## Actual Behavior:
Both the deleted Windows id_rsa.pub AND the custom key are present in authorized_keys

## Analysis:
While I understand the auto-detection of ~\.ssh\id_rsa.pub was implemented for user friendliness to improve first-time SSH success rates, the current behavior ignores explicit user intent when they delete the auto-populated key.

To investigate the root cause, I tested by modifying the first character of ~\.ssh\id_rsa.pub incrementally throughout the imaging process. This revealed that the imager scans and captures the Windows key at the moment "Save" is pressed, regardless of whether it was previously deleted from the UI. This is why I believe it might be a separate issue to #1160 

## Security Implications:
This creates an unintended attack surface - users believe their Pi is secured with only their custom key, but there's actually a second access method they're unaware of. If the Windows machine is pwned and attackers gain access to the private key at ~\.ssh\id_rsa, they would also have access to the Pi. Users who think they've isolated their Pi with a custom key securely stored elsewhere may not realise they need to rotate the Pi's authorized_keys after a Windows compromise.

### Version

1.9.6 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Windows 11 Pro 24H2

### Selected OS

Rasberry Pi OS Lite 64-bit

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

dwalton64, 2025-08-30:
I am seeing something similar.  I want to use a password but have a key from a previous version.  I check the password option but when the pi boots, it appears to use a key and I am not given an option to enter the password I set, even when I delete the default key that I previously used.

lwbeam, 2025-08-31:
I'm seeing the same problem with the Mac version (1.9.6) running on an M3 MacBook Pro. Imager auto populates authorized_key, even after it's been deleted and password authentication has been selected. Next time Imager is used, public-key authentication is applied, even if password authentication has been selected.

ngist, 2025-09-16:
Is there already an issue for step 6 "use scroll wheel while hovering over key rows to check none hiding with UI glitch"? I added a comment to #629 about this but didn't see a dedicated issue.