raspberrypi/rpi-imager #1189: [BUG]: Pre-configured WiFi via Raspberry Pi Imager does not take effect on trixie

jvonau, 2025-08-31:
### What happened?

Used the OS Customisation to embed the WiFi info but didn't auto connect on boot. Digging further showed that /usr/lib/raspberrypi-sys-mods/imager_custom embedded in firstboot.sh was not installed on the target image. See also https://github.com/RPi-Distro/raspberrypi-sys-mods/issues/102

### Version

1.9.6 (Default)

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

U24.04

### Selected OS

Raspi OS Trixie nightly https://downloads.raspberrypi.com/nightlies/ 

### Which Raspberry Pi Device are you using?

Raspberry Pi 5, 500, and Compute Modules 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-08-31:
> Raspi OS Trixie nightly

We're going to be changing the way that Raspberry Pi Imager interacts with Raspberry Pi OS Trixie, before we release our "stable" Trixie images.

caternuson, 2026-11-12:
This issue is closed, but I think there are still some potential edge cases that are still not working.

I am unable to connect to a hidden wifi using a Pi Zero 2W. More info here:
https://github.com/raspberrypi/trixie-feedback/issues/23#issuecomment-6829872096

Also not sure where the best repo is to raise this issue? Not sure if this is rpi-imager or trixie related. Or maybe both?