raspberrypi/rpi-imager #1205: [BUG]: Writing any image other than Rasbian takes an inordinate amount of time.

tlewis23, 2025-09-10:
### What happened?

Any image that is not Rasbian takes upwards of 2 hours to write to a NVME drive in a USB 3 enclousre. Rasbian, less than 2 minutes. Ubuntu, any flavor or version, 2 hours.

### Version

1.9.6 (Default)

### What host operating system were you using?

Windows

### Host OS Version

11 24H2

### Selected OS

Ubuntu Server 24.04 

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

USB SSD

### OS Customisation

- [x] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell

```

lurch, 2025-09-10:
I wonder if this might be a network / ISP issue? Could you identify the relevant URLs in https://downloads.raspberrypi.com/os_list_imagingutility_v4.json please and try downloading them in your web-browser, to see if that also results in the Ubuntu images downloading much slower than the Raspberry Pi OS images?

lurch, 2025-09-17:
I guess you need to get in touch with your ISP then?